vitotai/BrewManiacEsp8266 #127: Settings are not saved

Omnn82, 2024-03-31:
I have the following problem - no settings made via menu or web-interface are saved. I define sensors, set what and where to use, after that everything works and shows, but when I turn off the controller again goes to the initial screen, sensors are absent, my wi-fi point is also absent. I have to do it all over again.
I tried the following versions 0.5.1, 0.5.2, 0.5.3. My network finds, connects, ip is assigned - everything works, but as soon as you turn off - everything forgets.
Also noticed that after connecting to my access point, the network bm continues to be visible in the list of access points and you can connect to it.
I also failed to save the mashing settings via web-interface - it wrote failed to save.

Omnn82, 2024-04-02:
Bought a new nodemcu board, completely wipes all data. I flashed with already ready firmware from BrewManiacEx.multisensor.bin github to exclude that I compile something wrong.
Without connecting the screen and other contacts, I turn on, see the access point "bm", connect to it. I try to go to the "Automation" section in the web interface and make some changes, for example, change Mash in to 60°C and click save. The error Alert "fail to save!" appears.
I also tried connecting to a wifi point, everything works, IP is assigned, it connects. But if you turn off and then turn on everything is forgotten and you see the initial screen.
Help to solve the problem.
Thanks in advance.
[image]
[image]
[image]

vitotai, 2024-04-02:
Try format filesystem by browse to
http://[YOUR controller's iP or xxx.local]:8008/format-spiffs

If it doesn't work, check your system status by click the "BM" on the web page. You might also check the file on file system by browsing to
http://[YOUR controller's iP or xxx.local]:8008/filemanager

There might be something wrong with the flash, but I don't know what.

Omnn82, 2024-04-04:
Help me, I bought another type of board - a new board wemos d1 mini. I uploaded firmware.bin through NodeMCU Pyflasher (attached screenshot), everything flashed. I go to 192.168.4.1, click on BM and again I see File System Size = 0. I tried to change the value in automation - it says that it is impossible to save.
Is it possible that I'm flashing somehow wrong?
I tried flashing through Flash Download Tool, having pressed erase beforehand - the same thing.... 
I tried flashing on two PCs.

Translated with DeepL.com (free version)
[image]
[image]
[image]
[image]